Jumpy07/Solis---SolisCloud-and-Home-Assistant #1: More info on GC-1201K

jmccrohan, 2023-04-24:
Hi,

Thanks for putting this together. Adding the optoisolator is a clever evolution of @alienatedsec's design.

Can you share a bit more info on the GC-1201K please? Expected price, links to aliexpress etc? There seems to be a few different models available, at different price points. 

1. https://www.aliexpress.com/item/1005003466781251.html (Item: type11)
2. https://www.aliexpress.com/item/1005005128549371.html
3. https://www.aliexpress.com/item/4000389383921.html
4. https://www.aliexpress.com/item/1005004258214983.html
5. https://pt.aliexpress.com/item/1005002483681845.html

Are these devices bidirectional and therefore identical to their much cheaper _single master to dual slave_ cousins?

Thanks,
Jon

alienatedsec, 2023-04-25:
A great approach, indeed. Well done @Jumpy07, and congrats on your first attempt with GitHub :) We can collaborate more if you could enable `Discussions` for this project. Raising issues is likely to create much more noise.

@jmccrohan I looked at the same and have similar thoughts - links 4 and 5 seem the most relevant, but the others look like clones and could not have the same functionality. However, looking at @Jumpy07 photos, it looks like @Jumpy07 is using something different from what is listed.

The photo:
[image]

Listed item:
[image]

@Jumpy07 which one did you order? The link you posted on the Facebook group is also not reflecting what you installed.

alienatedsec, 2023-04-25:
> Sorry didnt realise there were some many clones / different models.. This is the one I used: https://www.aliexpress.com/item/1005005128549371.html?spm=a2g0o.productlist.main.1.353847f3QnMMn4&algo_pvid=60b48212-7e50-4f36-a683-2d5b3f50ac56&algo_exp_id=60b48212-7e50-4f36-a683-2d5b3f50ac56-0&pdp_npi=3%40dis%21GBP%2122.75%2120.93%21%21%21%21%21%40211bd8be16824105524221086d07da%2112000031771589334%21sea%21UK%21164589609&curPageLogUid=ZJza73apo6la I
> […](#)

Thanks @Jumpy07 

But still, it does not look like the one you got installed.

ciaranmcgirl, 2023-05-13:
Is that the exact shop/link you bought from @Jumpy07 ? The different labels on each device through out this thread doesn't inspire confidence but if it's the exact link/shop I'd be happy enough to go with that.

Jumpy07, 2023-05-15:
> Hi,
>
> Thanks for putting this together. Adding the optoisolator is a clever evolution of @alienatedsec's design.
>
> Can you share a bit more info on the GC-1201K please? Expected price, links to [...]

There are three similar models, the model that is required is the GC-1201K, have updated git with the link and a few links to the manfacturers website (in chinese)

As usual on Aliexpress some of the pictures dont match the item.. 

I reached out to the manufacturer to ask if there was an English version of the manual and app... but no response to date

Best Regards
Craig

closed 2023-05-15 by Jumpy07