Jumpy07/Solis---SolisCloud-and-Home-Assistant #2: EW11 missing IP address

Richard7619, 2023-06-11:
Hi,

Great Project.
I have the hardware installed and it appears to be working. 
However I’m having issues setting up the IP address. 
I can see EW11 as a Wi-Fi network so connect and go to 10.10.100.254
the status shows Wi-Fi disconnected and no IP address. 
The photos show the setting I’ve used. Are there any I need to change? 
Also how do I create a static IP address? 

Thanks Richard 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jumpy07, 2023-06-11:
Remember and hit submit when you change the settings.. (button at bottom)

 

Do a scan for wifi.. enter password for you SSID

 

I am using DHCP on my LAN… so ip below only used if it doesn’t find DHCP server..

 

 <https://user-images.githubusercontent.com/135166611/244952393-c6f47835-77cd-411b-9fbb-824fcae656d4.jpeg> 
 <https://user-images.githubusercontent.com/135166611/244952411-52b142e1-394a-4f3a-b141-8648e8bc5a7f.jpeg> 
 <https://user-images.githubusercontent.com/135166611/244952427-1540f1f8-b7d1-4258-afef-6705ee6bb8b8.jpeg> 
 <https://user-images.githubusercontent.com/135166611/244952432-cfacf777-6665-4224-9e6d-15d0d2e41b25.jpeg> 
 <https://user-images.githubusercontent.com/135166611/244952442-cd2c39af-fefc-43ab-92c7-4ba124fb2356.jpeg> 
 <https://user-images.githubusercontent.com/135166611/244952457-e445c16f-24b7-4ca4-8984-392fc9b6d151.jpeg> 
 <https://user-images.githubusercontent.com/135166611/244952472-6fffb554-87ad-4f4a-acaf-27dba52a8386.jpeg> 
 <https://user-images.githubusercontent.com/135166611/244952480-554fc7a7-f4cd-4c89-9ba5-5aa5f703d588.jpeg>

Richard7619, 2023-06-11:
Thanks, I had been hitting submit and just connected again and settings in photos have saved including Wi-Fi and password. From the photos any idea what setting might be wrong?

Jumpy07, 2023-06-11:
Not really.. do you have DHCP turned off ?

Richard7619, 2023-06-11:
No DHCP is on.

Richard7619, 2023-06-20:
OK, So I think I have this part sorted. 
I'll explain in case other Newbies find it helpful. 
I was expecting the router devices to have a name like EW11 or Elfin but it's actually called lwip0.
That is connected so I can get it's IP in 192.168 format. This was my missing link. 
Then the settings page use that for the WAN IP, then 255.255.255.0 and the router IP for the gateway.
The Lan IP is the 10.10.100.254 which is the AP you use for setting it initially. 
The settings show it's wi-fi is connected and I'm showing some sent and received frames. 
I'm still having issues with the Solax integration but I'll ask separately on that related project. 
Thanks,

closed 2023-06-20 by Richard7619