Jumpy07/Solis---SolisCloud-and-Home-Assistant #3: Almost but not quite connected to Solax modbus integration

Richard7619, 2023-06-21:
I thought I had this but not quite. A little more help and hopefully this Newbie can do a happy dance. 

I seem to have my Erfin connected. I've filled out the settings in the Solix Inverter Modbus integration. Each time I hit reload the sent frames increases by 1 and the sent bytes increase by 8 (on EW11 status) in serial port, and the received frames increases by 1 and the received bytes increases by 12 in 'netp' 
So there seems to be some communication but the Hub states "no devices or entities"

I asked on the Solax project Q&A but was pointed back here which is fair enough: "there is no connection between HA and your Modbus interface. It's either port, protocol or IP wrongly configured." It's hard to advise as I don't use that solution. It would be best if you asked for connectivity and configuration help at the project's source that you have followed."

I run the debug log and get these

KeyError: 'value'
2023-06-21 12:02:56.581 DEBUG (MainThread) [custom_components.solax_modbus] using pymodbus library 3.x
2023-06-21 12:02:56.677 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2023 6
2023-06-21 12:02:56.678 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2023-06-21 12:02:56.678 DEBUG (MainThread) [custom_components.solax_modbus] setup entries - data: {}, options: {'name': 'Solis', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solis', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False, 'host': '192.168.1.218', 'port': 502, 'tcp_type': 'tcp', 'read_serial_port': '/dev/ttyUSB0', 'baudrate': '9600'}
2023-06-21 12:02:56.678 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-06-21 12:02:56.705 DEBUG (MainThread) [custom_components.solax_modbus] Setup solax_modbus.Solis
2023-06-21 12:02:56.705 DEBUG (MainThread) [custom_components.solax_modbus] solax serial port /dev/ttyUSB0 interface tcp
2023-06-21 12:02:56.705 DEBUG (MainThread) [custom_components.solax_modbus] solax modbushub creation with interface tcp baudrate (only for serial): 9600
2023-06-21 12:02:56.706 DEBUG (MainThread) [custom_components.solax_modbus] Solis: ready to call plugin to determine inverter type
2023-06-21 12:02:56.706 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Solis: trying to determine inverter type
2023-06-21 12:02:59.782 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.1.218, 502) failed: [Errno 113] Host is unreachable
2023-06-21 12:02:59.801 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis: attempt to read serialnumber failed at 0x80ec
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 53, in _read_serialnr
    inverter_data = hub.read_input_registers(unit=hub._modbus_addr, address=address, count=8)

It's probably something simple that is wrong but both seem to have the same IP and port 502 so I'm struggling. 
Thanks in advance

alienatedsec, 2023-06-21:
> I asked on the Solax project Q&A but was pointed back here which is fair enough: "there is no connection between HA and your Modbus interface. It's either port, protocol or IP wrongly configured." [...]

@Richard7619 I pointed you back because @Jumpy07 is the best to address your connectivity problem. However, I just had a second thought and while connectivity might be there, the cabling between your WiFi device and the HUB could be wrong. Make sure A and B are connected correctly. 

Is your data logger working correctly?

Richard7619, 2023-06-21:
I've just done my happy dance! 
1 device and 143 entities now connected. 
yes- the final issue was wiring. 
Thanks,

closed 2023-06-21 by Richard7619