Jumpy07/Solis---SolisCloud-and-Home-Assistant #4: Alternative

0rangutan, 2023-11-01:
Would this work as a cheaper alternative RS485 repeater?
https://www.amazon.co.uk/gp/product/B07X2GM3VY

hippytyre, 2023-11-25:
I hope so, I'm building my set-up tomorrow using one.

0rangutan, 2023-11-26:
Good luck - please share how you get on.

hippytyre, 2023-12-04:
> Good luck - please share how you get on.

It's not quite as neat as I would have liked but I'll go back and neaten it up. It works perfectly though.
[image]

hippytyre, 2023-12-24:
All I can tell you is that I've had no issues since I wired it all up. It all works perfectly

Jumpy07, 2024-03-25:
Software is outwith my Github project..which is purely hardware....hence have not responded.....Sent from Outlook for Android

ececconi99, 2024-03-26:
> Software is outwith my Github project..

not sure what you mean by "outwith"

closed 2024-04-21 by Jumpy07