Jumpy07/Solis---SolisCloud-and-Home-Assistant #10: Done the hardware now connection issues...

ececconi99, 2024-02-29:
I've logged into the ELFINEW11 hotspot and changed everything over to my home guest wifi network. I can access ii's web page using the new credentials. I also entered the ip address and port into the HA Solax integration configuration gui. HA says success but then shows no devices. 

It occurs to that although I entered ip and port, how can there be any communication when there is nowhere to enter my guest wifi network name and password? What am I missing here?

ececconi99, 2024-02-29:
Yes I’ve done that in the Elfin setup web page,  but how does Home Assistant / Solax integration get to know about my guest wifi login?

Jumpy07, 2024-02-29:
Dont undestand the question...why are you using guest wifi.....is HA on the same subnet as the ElFIN ?Sent from Outlook for Android

ececconi99, 2024-03-01:
I'd really appreciate some more help on this, please. I've built the hardware and can login to the ELFIN hotspot and can change the web page access credentials. I think I can also change the SSID and password, but I have no idea how to change the WAN , LAN, STA or AP configurations. Your GITHUB page doesn't say anything about how to set up those fields. AFAICS, you only show the comms settings page. Jevjev's YT doesn't go into that either.

Jumpy07, 2024-03-01:
This is my config…  WAN set your DNS server.. which will be your home router IP normally.

 

Set a static IP in LAN in case you loose DHCP.. and also set DHCP on..

 

Set Wifi as STA and add your SSID and Key

Once you have changed the settings don’t forget to hit save at bottom..  reboot the ELFIN

 

Use an IP scanner to find the ELFIN IP on the network…  or check your DHCP table in your router..   you can take a note of the MAC address of ELFIN which will help you identify it on your network..

ececconi99, 2024-03-02:
> This is my config… WAN set your DNS server.. which will be your home router IP normally. Set a static IP in LAN in case you loose DHCP.. and also set DHCP on.. Set Wifi as STA and add your SSID and [...]
Thanks for the reply. 
Can you clarify the WAN settings, please?
There are 5 fields:
DCHP - on or off?
WAN IP - is this my router ip address?
Subnet Mask 255.255.255.0 
Gateway - ? (this is where I had put my router IP address - is that wrong?)
DNS - ? 

As you can tell, I don't understand why a WAN section is needed. I thought one logged into the hotspot, changed the default login credentials, changed the device IP and and changed its WIFI ssid and password. From then on the ELFIN should be on my network. Clearly not so and there is more to this than I understand. Sorry.

ececconi99, 2024-03-02:
Update: I think I have the measure of it now. Many thanks for all your help and patience.

Your last response implied that only my router IP address was required for the WAN DNS entry and this was the only field visible if DHCP was left ON. So that s what I did. 
You said to leave DHCP on in the LAN setting too. This seems to result in the device ignoring what I had entered in the LAN IP. I could not access the browser page at the IP address I had set and nor could I ping it. 
So, I went looking in my router DHCP table and found it had given the ELFIN the address 192.168.0.91  So I set that as the fixed IP in my router and all was well from then on in. I'll tap in beteween the inverter and dongle as soon as I get mains for the power adapter in the loft sorted.

ececconi99, 2024-03-03:
Connected it all today. Ran the Solax integration using MODBUS address 1 (a guess), Solis inverter,  EW11 IP address and port. HA Solax said "success" and "initialising" but ended with no devices no entities.  I emailed Solis support for the miodbus number (I could not find it on soliscloud or my inverter webpage). 
Meanwhile, I tried various modbus numbers and worked my way down the SOLAX integration inverter list starting at Growatt. No dice until I tried SRNE. Success! HA SOLAX int now reports 1 device and 24 entities.
So FYI, the Solis inverter congfig in the SOLAX integration does not seem to work. I will switch over to asking questions of   wills106/homeassistant-solax-modbus
If you have any thoughts on this, I'd love to know.

ececconi99, 2024-03-03:
> Its working fine for many people...so must be your config......do you see
> any data in the EW11 status screen ?...Sent from Outlook for Android

No, not at all. All zeros on the status page  Serial Port State and
.Communication State - 'netp'  sections.  I must be missing something.
Maybe a wire has come loose.  I'll keep digging

ececconi99, 2024-03-04:
Checked the wiring - no fault found. The Solis cloud is working fine so the first RS485 feed through is working. The second one going to the EW11 measures ok continuity-wise. The EW11 ethernet light is on and flashes occasionally. The EW11 browser page still has only zeros in Serial Port and communication State sections.
 I'm out of ideas for further checking. Can you suggest anything?

Jumpy07, 2024-03-04:
Check you have A-A and B-B

ececconi99, 2024-03-05:
I checked and I had  A-A and B-B. Still no data in the EW11 status screen.

So... I swapped the wires around A-B and B-A. Lights started flashing and I got data in the EW11 status screen! I guess the GC-1201 and EW11 manufacturers uses a different naming convention.

Back to Home assistant and I deleted and re-installed the SOLAX Modbus integration. Then I configured it, using slave/master, port 502 and "solis". It took much longer to initialize but then still nothing; no devices, no entities. 

Can you confirm that slave/master, port 502 and "solis" are the correct HA SOLAX integr configuration parameters, please?

ececconi99, 2024-03-05:
I guess I can close this now that the hardware seems to be working.  I'll transfer over to wills106/homeassistant-solax-modbus to try to sort out the HACS Solax Modbus config.

Many thanks for all your help and patience.

I think you and JevJev should consider mentioning that RS485 naming conventions can be a bit flakey and that trial and error may be necessary. Your wiring diagram may not apply.

ececconi99, 2024-03-07:
I've had two replies via facebook and YT telling me "You can't split the rs485 out and use the ha solax modbus addon. It clearly states in the manual for the add on its needs (sic) sole access or the rs485 port." 
I assume they are wrong and the GC-1201 RS485 / Modbus Dual Host Adapter gets round this. Can you confirm this?

Jumpy07, 2024-03-07:
Did you read what the GC-1201 does ?   clearly states two masters... 
Did you read the Github page ?

RS485 and Modbus are a single master multiple slave in theory.. hence 
why you need the GC-1201...      otherwise I would have wasted my time 
writing the Github entry...

There are other solutions that you are free to Google as well...

Jumpy07, 2024-04-21:
Resolved

closed 2024-04-21 by Jumpy07