Jumpy07/Solis---SolisCloud-and-Home-Assistant #11: Backup Restore Method missing - page 404

ececconi99, 2024-03-11:
I found this page https://github.com/LucidityCrash/Solis-Modbus-Auto-Charging-using-Solax?tab=readme-ov-file
whic seems to have all the code needed to get Home Assistant to talk to a Solis inverter via your 2 master hub + Elfin wifi

Not being a HA or YAML expert I was particularly pleased to see the Backup Restore Method link here
https://github.com/jevburchell/Solis-Modbus-Auto-Charging-using-Solax/blob/main/HA%20Backup%20Restore.md

Sadly the link does not work. It produces Page 404. 

As Jevjev is not currently replying to comments on his YT video and you seem to have colaborated with him on this project, could you enquire as to what the situation is. with a view to getting the link working again. please?

ececconi99, 2024-03-15:
In the absence of the BAckup/restore method, I have installed all the YAML code from this page https://github.com/LucidityCrash/Solis-Modbus-Auto-Charging-using-Solax?tab=readme-ov-file successfully.
I now have the full dashboard running but with warnings of entities not found. I have resolved many of them by going through the code and changing Solax to Solis, as I have a Solis inverter and had used the Solis version of the Solax Modbus integration config.
I'm still missing a few steps, e.g. "number.solis_timed_charge_current Entity not found" and similar relating to the charge start and end times. 
I'm aslo a bit puzzled as to how this setup controls the battery charging as there seem to be no mention of  commands going back to the inverter via Modbus, MQQT or RS485, etc. 
Can you shed any light on this.

chronvas, 2024-04-18:
The link points to a (probably forked) repository by the user https://github.com/jevburchell/ which no longer exists, maybe the user deleted their forked repository.
You can use the equivalent link of the repository by the user https://github.com/LucidityCrash (on your first link) and it looks similar to this https://github.com/LucidityCrash/Solis-Modbus-Auto-Charging-using-Solax/blob/main/HA%20Backup%20Restore.md
Or the link from the repository from the user https://github.com/fskelly which user https://github.com/LucidityCrash forked from, and it is probably this https://github.com/fskelly/Solis-Modbus-Auto-Charging-using-Solax/blob/main/HA%20Backup%20Restore.md

On your second question, I have the same question and I started a **discussion** on the responsible [integration](https://github.com/wills106/homeassistant-solax-modbus/discussions/844) page, because when I try to setup charge times via the solax-modbus integration nothing happens. Other settings seem to work like "force charge SOC" but not this one in the discussion.

Jumpy07, 2024-04-21:
Sorry this is outside the scope of this repository

closed 2024-04-21 by Jumpy07