Jumpy07/Solis---SolisCloud-and-Home-Assistant #12: Help with build

cfarm31, 2024-05-25:
Afternoon all,

first time builder so please be kind. I got my rs-483 hub and elfin the other day finally from Ebay and put it all together as per the wiring diagram and hooked it all up and then when running the datalogger connected fine there was a red light for power and nothing else lit up which i think was a problem 

As i say the datalogger worked fine but when trying to add the solax integration i kept getting errors about  serial number unknown. I tried alot of things including double checking my wiring but could not a green light on the connection until i swapped round the wires for the elfin and the green light came on but i still got the same errors and the system just would not work 

Any help from anyone would be really appreciated because i cant think of what is wrong i checked everything multiple times and just couldn't get it working

Jumpy07, 2024-05-25:
Sounds like wiring issue… you need to double check 

 

No idea what serial number unknown is I am afraid

fastsonic, 2024-10-01:
Hi, when its say serial number onknown i think you have to send the Serialnumber to the Solax guys. its not a Problem from here.